bromagosa/Snap4Arduino #280: I would like to acquire temperature data and upload them to my computer

IlundainIlundain, 2019-05-15:
How is it done by the serial port? Can I export them as a file? I want to leave the temperature sensor taking data and then analyze them, how can I do it? Thank you

jguille2, 2019-05-15:
Hi, @IlundainIlundain ,
Every data you can read from Snap4Arduino can be stored (variables, lists...). After this, that data will be stored in your project. If you want, you also can export "only the data" to use it with other apps (spreadsheets...).

For example. With...
[image]
you will get something like
[image]
and "right click" over this variable watcher...
[image]
you will get an CSV file to use that data.

Of course, you also can play with that data inside Snap4Arduino, playing with that list: calculating, graphing...

Joan

IlundainIlundain, 2019-05-15:
COOL. The information is concrete and simple. THANK YOU. Surely it will be
a very silly question, I'm starting with this program where do I find data?
THANK YOU

jguille2, 2019-05-15:
Hi!
I'm not sure if the question is about the "data" element?

In the "Variables" category, at the beginning, you have the "Make a variable" action. In that example, I created a "data" variable.

After this, the variable blocks (like "set () to ()")  will have that variable name into their dropdown menu.

You will see that new variables have a checkbox beside, to show/hide their values inside the stage. This "values on the stage", called "variable watchers" has that "right click action" to export their content.

Joan

aletricks, 2021-07-16:
How can I use list in case of translatable project?

jguille2, 2021-07-18:
Hi @aletricks ,
Uf, this is another thread, isn't it? Maybe I can close it... but I want to answer before... although this hasn't a simple answer...

1. First of all. The "transpile_to_Arduino feature" is only a small patch, to allow the creation of simple Arduino programs that can work autonomously (by uploading them to the board). And nothing more.

   And this "simplicity" has not been for lack of time ... but because it invites confusion (this is not the line of Snap4Arduino, where we want to enjoy the dynamic and lively environment that Snap! is offering us) and also because it does not have sense (I can't translate Snap! programs to Arduino in general) 

   This "loss in translation" is due to many factors. In one hand we have a lot of features that Arduino have not ( costumes, sounds, sprites, a screen to draw and stamp...). We also have important differences that cannot be translated directly (non typed variables, linked lists...). And really, if you code in the Snap! way, you will have a lot of untranslatable issues (parallelism, recursion, first class elements, high order functions, hyperblocks...)

2. But at the same time, this  "transpile_to_Arduino feature" is a sample of the _codification_ feature that you have in Snap! And yes, even it is not a Snap! goal, you can use Snap! environment experimenting with codification and using blocks as a visual representation of a text-based programming language.

   Just right click on any block in the palette (when you are in a "translatable project" of also you can got to the _settings menu_ and select "Codification support" on any Snap! project) and you will see two interesting options: "header mapping..." and "code mapping..." And with them, you can create your own translation feature, and also add translatable blocks. Note that you can "show primitives" to see the other blocks... and also you can create custom blocks and use with them this codification feature.

3. Note that if you want to support "lists" (or other features) you will have to do some work... because as I said, Snap! isn't translatable to Arduino... and then, you will need to take some decisions and "code" simpler solutions. And then, to use this with Arduino, you will have to take care with your projects, because you will have to code in the Arduino way, not using all Snap! features with lists!

4. Some examples to help you... but as I said, it needs extra work, and just define what do you want (not all Snap! features) and converting them to Arduino code...

[image]

[image]

[image]

and then... this is running...

[image]

That's all!
Joan

closed 2021-07-18 by jguille2