bromagosa/Snap4Arduino #281: ACCELEROMETER + GIRÓSCOPE MPU-6050 I2C communication

IlundainIlundain, 2019-05-15:
I would like to use this Arduino component with program, but I do not know how to do it, where can I get a library? How is it added? Are there web pages where there are more librarians? Thank you

jguille2, 2019-05-15:
Hi @IlundainIlundain ,
On the board site, you don't need libraries, because Firmata supports I2C connection. So Snap4Arduino is able to play with this sensor.

But, this is not direct!

Inside the Snap4Arduino, you can load the "i2c" library. And you will get the "i2c basic block" to do this job. But there is quite work to get the results you want... Reading board specifications o reading other implementations (like this [https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050)) you can do this work.

I would like to help you in more detail, but I do not have any sensor like this to test.

Joan

IlundainIlundain, 2019-05-15:
Thank you very much for the response and your time. I would like to make
a drone with my students and I need something similar to an accelerometer /
giroscope  will not you know something similar? I know how to do it in
Arduino but not with a program based on Scratch (S4A, mBot ...) For kids, I
would like to use a program based on Scratch. THANK YOU

wandiufo, 2022-10-15:
Now  I am working to use i2c MPU6050. I have tried using available  snap block for i2c but not success, then I put the simple worked MPU6050 to firmata firmware. I can get a byte from snap4arduino but I have difficulties to get array of byte or integer (2*byte) form JS snap block. Could you inform me how to pass array or integer from arduino firmata to snap?

Best regards
Irwandi

part of code for firmata
case 0x06:
          readFrom2(MPU6050_DEVICE,0x6B,6);  //fill buff
          i2cRxData[0] = MPU6050_DEVICE);
          i2cRxData[1] = 0x6B;
          numBytes=6;
          //for (int i = 0; i < numBytes && Wire.available(); i++) {
          for (int i = 0; i < 6; i++) {            
            i2cRxData[2 + i] = buff[i];
          }

          // send slave address, register and received bytes
          Firmata.sendSysex(SYSEX_I2C_REPLY, numBytes + 2, i2cRxData);

jguille2, 2022-10-18:
Hi Irwandi,

Maybe the LCD example at the other issue (https://github.com/bromagosa/Snap4Arduino/issues/294#issuecomment-1281680434) helps you to test i2c devices (in general)

Test that before and we can continue later here. Continue...

Joan

closed 2023-05-25 by jguille2