bromagosa/Snap4Arduino #288: Send Sysex message from arduino to snap4arduino

deno750, 2019-08-01:
Hello, I'm trying to figure out how to receive data from arduino throgh sysex. On my firmata code I did this:

`Firmata.write(START_SYSEX);
      Firmata.write(STRING_DATA);
      Firmata.write(2);
      Firmata.write(END_SYSEX);` 

And on the block definition I did that:

`this.arduino.board.once('string', function (data) { ready.add(data); });
this.arduino.board.sp.write(new Buffer([0XF0, 0x0D, 0xF7]));`

Here there's the entire block definition.
[image]

When I try to run the block, I don't receive any value from arduino.

I'm using snap4arduino through ecraft2learn. The chromium/chrome plugin is intsalled correctly and the arduino connects successfully.

jguille2, 2019-08-29:
Hi Denis (@deno750)!

Excuse our delay, but holidays... :(

I have to test this to be sure, but some comments to start...

I need more info. I guess you have defined a new SYSEX command, haven't you? Because your "firmata code" must be into a `case 0x0D:` section (you use this '0x0D' to run the command from the block).

I you have already done this, test tho change "this.arduino.sp.write" to "this.arduino.transport.write".

Test this, and we continuous....

Joan

lhong01, 2022-03-02:
Hello @deno750 sorry to bother you. I'am a newbie with s4a and wanted to ask you where i can find information in order to program my own s4a blocks with javascript

I googled a lot about this.arduino.board function but i'am not finding anything...and a little about sysex codes...

This just helped me a little bit
https://blog.s4a.cat/2015/03/13/Extending-Firmata-for-Snap4Arduino.html

jguille2, 2022-03-02:
Hello @lhong01 

That doc (from Citilab) is fine.

If you want to see more examples, maybe it is interesting for you [SA5Firmata](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md)
- Check the SYSEX commands inside that Firmata
https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata_ir.ino#L689-L1041

- Their blocks are [here](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata_ir.xml)
But, easier to check, use http://snap4arduino.rocks/v6/ and load 'SA5FirmataIR' library.
I'm pointing to version 6, because in current version JS code is "hiden" into primitives.... It's the same, but we need it now to avoid the need to allow (checking) "Javascript extensions".

Tell me if you need something more...

Joan

closed 2022-03-02 by jguille2