bromagosa/Snap4Arduino #289: Concurrency Issue?

PJ789, 2019-08-19:
Noticed while using Snap 4 Arduino 1.2.7... if I interact with a single Uno from event driven code blocks in two or more sprites... it gets very confused and often becomes unusable/unresponsive.

I guess S4A missing some kind of mutex to serialize concurrent access to the Arduino Firmata code from multiple sprites.

PJ789, 2019-08-20:
Example of the problem attached. Two sprites, each attempting to control 1 of 2 LEDs by continuously flipping the LED state from on to off.
Results in error messages.
[issue_289_example.zip](https://github.com/bromagosa/Snap4Arduino/files/3518612/issue_289_example.zip)

PJ789, 2019-08-20:
To add… OS is Windows 10. s4A is 1.2.7. Arduino IDE is 1.8.9. Arduino is an Uno clone with an FT232 connected to COM14.

PJ789, 2019-08-20:
Here's a slightly better example.

I have two simple code blocks each of which control 1 of two LEDs (as prev).

If I connect the Arduino manually, and run the code, one of the two blocks will cause a confusing error message to appear  "Inside: Error Arduino not connected".

[issue_289_example2.zip](https://github.com/bromagosa/Snap4Arduino/files/3518684/issue_289_example2.zip)
[image]

PJ789, 2019-08-20:
Am I right in thinking... one Arduino can only be connected to/controlled by one sprite?
Therefore, any custom code block that operates in the context of a second sprite that is not connected to the Arduino won't work?
Obviously, that restriction could make it a little difficult to create tidy code that controls multiple devices via a single Arduino interface (.... without also modelling the Arduino itself as a controller sprite perhaps).

bromagosa, 2019-08-20:
Exactly. An Arduino board is a property of a single sprite.

If you want another sprite to interact with the Arduino board that another sprite has connected, you need to use "ask sprite to do", from the Control category, or broadcast messages.

PJ789, 2019-08-20:
Thank you.

I love what you've done with Snap4Arduino. Good luck with your launch in September.

I'm running a childrens workshop at the beginning of October… and plan to use Snap4Arduino (with Nanos).

closed 2019-08-20 by PJ789