bromagosa/Snap4Arduino #293: Snap4Arduino cannot connect to Arduino

Nikau, 2020-01-26:
Firmata is loaded and tested on my Arduino but SNAP4Arduino either cannot find board or connects and then within seconds disconnects. I have tried with both the downloadable version and the online version on Chrome with relevant installed extension.

jguille2, 2020-01-26:
Hi @Nikau,

Chromium plugin should work too, but let's look at the desktop version first, that is the most stable and complete.
Many possibilities here... but you have said sometimes it connects and disconnects after few seconds. So, I guess drivers, port permissions, firmware... is ok. My question, What board have you? And what elements (motors, servos, bluetooth...) have you connected to that board?

Sometimes the problem is only on power supply. If your elements consume too much intensity, maybe the communication (PC - board) is broken (maybe only in small moments) and then, the board is disconnected from Snap4Arduino.

Test with your board without any element or change the power supply.
Continue...

Joan

Nikau, 2020-01-27:
Hi,
thanks for getting back to me...the board I am using is based on the
Arduino Pro Board(The Victoria University Wellington(New Zealand)
Quizmaster board is based on an Arduino Pro board (
arduino.cc/en/Main/ArduinoBoardPro), which uses an ATmega328
microcontroller.)I successfully uploaded the Standard Fermata sketch and I
have checked to see that this has loaded and it all checks out. Please note
that there are no external devices connected to the Arduino Board. The
desktop version of Snap4Arduino sometimes connects with the board
successfully but moments later disconnects. I will try again tomorrow with
the Arduino Nano. I am really wanting to use this package with my high
school students to help them learn about Robotics. Regards, John Bleakley

closed 2023-05-25 by jguille2