bromagosa/Snap4Arduino #294: LCD 20x4 on I2C not found

a3areval, 2020-04-16:
I want to use this type of display with Snap4Arduino. I was able to find instructions for a different type of LCD display back from 2015. Is there a tutorial that I can follow in order to generate this type of interface for Snap4Arduino for different devices?

bromagosa, 2020-04-17:
Hi Andres. I replied to you on YouTube already, but here's the same response just in case someone else needs this in the future:

The instructions for modifying Firmata are still be up in the old S4A blog. Here's a link to the article: https://blog.s4a.cat/2015/03/13/Extending-Firmata-for-Snap4Arduino.html

Between this article and the LCD one you should be able to manage. Let me know if you run into trouble!

jguille2, 2020-04-17:
Hi @a3areval ,
As Bernat has said, modifying Firmata would be the best solution, for better performance (speed) and to take directly advantage of the Arduino libraries.

But I want to note that your device is I2C, and Snap4Arduino has already a I2C library. You can play directly with this. But yes, you will have a "lower level" control of that device and, then, you will need to learn what Arduino library is doing. And, in the end, you'll get a slow LCD.

So, playing with I2c blocks are not a good maker solution... but yes, it can be a good learning one!

I had done a demo, some years ago... I will try to recover, and share here...

Joan

wandiufo, 2022-10-13:
Hi @jguille2 
Can you share the demo for LCD with I2C?
Irwandi

wandiufo, 2022-10-14:
[image]
I have tried, but not success, could you share it
Best regards
Irwandi

wandiufo, 2022-10-15:
[image]

I have succeeded direct modified firmware using sysex and modified snap block. Using the I2C block too difficult and not success. Currently I want to transfer array of byte from i2c device, but I have difficulty how snap block get array of byte or integer (2*byte) sending by sysex.
for example:
      case 0x06:
          readFrom2();  //fill buff
          i2cRxData[0] = MPU6050_DEVICE);
          i2cRxData[1] = 0x6B;
          numBytes=6;
          //for (int i = 0; i < numBytes && Wire.available(); i++) {
          for (int i = 0; i < 6; i++) {            
            i2cRxData[2 + i] = buff[i];
          }

          // send slave address, register and received bytes
          Firmata.sendSysex(SYSEX_I2C_REPLY, numBytes + 2, i2cRxData);
          break;

it is not work
[image]

Manual or information how  snap block get array of byte or integer (2*byte) sending by sysex (firmata) and read by snap (javascript) is important for me. 

Best regards,
Irwandi

jguille2, 2022-10-18:
Hi Irwandi,

I've made a test project to show you the direct i2c option.
You can test it here (https://snap4arduino.rocks/run/#present:Username=jguille2&ProjectName=lcdi2cSEC5) . It runs on Snap4Arduino desktop app and also on online versions (with Chrome/Chromium and with the Snap4Arduino-connector extension).

It's easy to implement other features like "cursor", "blink"... 

Test if you want... and we can continue.

I have never given importance to LCD libraries, because in Snap4Arduino we have always the computer screen... and you know, with many more features. But maybe it is interesting if you play and create interactive objects.

Joan

wandiufo, 2022-11-20:
Dear Joan

Thank you for you sharing file, it work well. The code only use i2c send command without read and it take longer time to display then direct modification firmware. It take long time because many wait 0.01 secs. 
Irwandi

jguille2, 2022-11-21:
Hi Irwandi (@wandiufo )

Yes, it takes longer times, I know. It was [my comment here](https://github.com/bromagosa/Snap4Arduino/issues/294#issuecomment-615103251).
If you want to build "scrolls" or something like this, it's too long, I think.

You can test, but without those "waits", some "strange thing" can happen... because we are sending commands one after one (just the same core libs do), but we are sending by serial... and if you try to do faster, you will get issues (strange chars...)
But maybe you can try slightly smaller waits... i was conservative (to make it work).

Again: better performance and speed with the modified Firmata... but for occasional use, it's a good option.

Joan

closed 2022-11-21 by jguille2