bromagosa/Snap4Arduino #299: Add Arduino blocks as a library in Snap!

Dan-TEALS, 2020-07-31:
Snap4Arduino seems to be falling behind the current release of Snap! Could it be added to Snap! as a library, like the Hummingbird robot and Leap controllers have been? It would be helpful to have Arduino support in the main Snap! branch.

jguille2, 2020-07-31:
Hi @Dan-TEALS,
Snap4Arduino team have been also working for the Snap! release.
And soon we will release Snap4Arduino6 (with our Snap!6 updated).

Your request would be great, but not possible.
Snap4Arduino has a serial connection, and also a web-server. And these features can't be inside a Snap! library. Browsers can't connect directly to PC serial devices.

Other approaches (I guess like Hummingbird, I'm not sure) use a middleware (another software) to connect to devices. Then, they are not really "only a Snap! library", because they need these other software running.

We have studied (and tested) other possibilities (other soft by websockets connection, chrome/chromium extensions, new (experimental) web serial protocol...) And maybe we will change... But now, our desktop version (built with node.js and using firmata serial connection to our devices) is good, fast and stable.

To do a more complex electronic stuff (controlling time to microseconds, autonomous projects...) we use Microblocks project (connected to Snap! by networks protocols). But we continue to enjoy with our Snap4Arduino features, playing with a lot of kinds of devices (Arduinos, ESPs..s)

Stay tuned and we will announce the next version soon (with the full Snap! 6 running inside!!)

Joan

Dan-TEALS, 2020-07-31:
Sounds awesome!

Thanks for the update, Joan. What you said makes sense and I'll admit to not being familiar enough with how Snap! handles libraries, which is why I asked the question. Scratch seems to deal with this by using browser extensions instead of "libraries," which does give access to the USB port for programming micro:bits, for example.

I'm looking forward to Snap4Arduino6. This is a really cool project!

Dan

Dan Stormont (he, him)

dan@tealsk12.org | 520.415.6206

Regional Manager, Arizona | Concordis, LLC

Microsoft Philanthropies TEALS Program

Computer science in every high school

microsoft.com/teals<https://www.microsoft.com/en-us/teals>

Microsoft respects your privacy.  Review our online Privacy Statement<https://privacy.microsoft.com/en-us/privacystatement>.

Microsoft Corporation

One Microsoft Way

Redmond, WA, USA 98052

closed 2022-12-27 by jguille2