bromagosa/Snap4Arduino #304: "Application Damaged" - Snap4Arduino_desktop-osx-64_6.2 - Mac OSX 10.12.6

RRWagner, 2020-10-15:
Trying to open the Snap4Arduino_desktop-osx-64_6.2  app with Mac OSX 10.12.6 gives:

[image]

This looks similar to this error/cause with a Mac OS X installer:

https://osxdaily.com/2019/10/24/fix-install-macos-application-damaged-cant-be-used-error-mac/

jguille2, 2020-10-16:
Hi @RRWagner ,

I'm sorry. I will try to check this... but it is strange.
Commenting...
- Snap4Arduino developers haven't got any Mac device. So it is difficult for us to see these issues.
- But before releasing, we always ask friends to test. And they said OK (also for 6.2)
- I see we have more than 200 downloads of the MAC version, and this is the first report about this. It is strange.

As I've said, I will check this (asking friends) but please, if you can check it again. Or if you can check it with other devices, OS versions...
Continue...

Joan

Surferwolfe, 2020-10-24:
I am having the same problem with my Mac Ver. 10.15.7

maprzybylla, 2020-12-16:
Yes, it's true, the problem occurs - but there is a work around: click cancel, go to system preferences → Security and Privacy → General. There you should find magically appearing dialogue that allows you to install the software anyway. It's weird that the option is not given in the pop up window, but it still is in the system settings.

HyperDuino, 2020-12-16:
In OS X 10.12.6 the error is "damaged", and not one that says anything about security.

[image]

following that error, system preferences → Security and Privacy → General, does not display anything about the app not opening, as is usually the case with programs that don't have the signed developer data. 

[image]

maprzybylla, 2020-12-16:
I'm working with OSX Big Sur (11.1) and used Snap4Arduino on all previous versions for the last few years. I'll redo the installation process and see what happens on another device - will notify shortly.

maprzybylla, 2020-12-16:
Ok, I tried it and get a different error message - for me it works as described above:

-  I get the error message _"Snap4Arduino" cannot be opened because the developer cannot be verified._
- then I click cancel
- then I get the option to open the App anyway

bromagosa, 2020-12-16:
Thank you, @maprzybylla! Glad to know we still run on Mac... :worried:

RRWagner, 2021-02-03:
Still no changes here.

[image]

[image]

[image]

I don't get the option to open the app anyway. I have seen that on other apps, but not with this file.

edutec, 2021-02-15:
Maybe this can help...

https://techsviewer.com/allow-installation-of-apps-from-anywhere-in-macos-mojave/

Regards.

-- 

*José García*
Coordinador àrea formació
+34 93 551 06 60 (ext. 208)
jogaye@e-citilab.eu

FUNDACIÓ PEL FOMENT DE LA

SOCIETAT DEL CONEIXEMENT

CORNELLÀ

Citilab Cornellà | http://citilab.eu | Pl. Can Suris s/n | 08940 Cornellà
de Llobregat | Barcelona

En compliment de la legislació vigent en matèria de Protecció de Dades de
Caràcter Personal, l’informem que les seves dades de contacte seran
tractades per FUNDACIÓ PEL FOMENT DE LA SOCIETAT DEL CONEIXEMENT (CITILAB),
amb la finalitat de mantenir la relació amb els usuaris dels diferents
oferts, inscripció als cursos i a les activitats organitzades, i en el cas
de que vostè ho hagi sol·licitat, l’enviament de publicitat relativa als
serveis oferts. FUNDACIÓ PEL FOMENT DE LA SOCIETAT DEL CONEIXEMENT
(CITILAB) assegura la confidencialitat de les dades aportades i garanteix
que, en cap cas, seran cedides per cap altre ús sense que hi hagi un
consentiment previ i exprés per part seva.

Per exercir els drets previstos en la llei sobre les seves dades, pot
adreçar-se mitjançant correu postal a FUNDACIÓ PEL FOMENT DE LA SOCIETAT
DEL CONEIXEMENT (CITILAB), Ref. Protecció de Dades, Pl. Can Suris S/N,
08940, Cornellà de Llobregat, Barcelona, o bé exercir els drets assenyalats
mitjançant l’enviament de la seva petició al correu electrònic
avislegal@citilab.eu, indicant en l’assumpte del correu “Ref. Protecció de
Dades” (acreditant degudament la seva identitat).

closed 2022-12-25 by jguille2