bromagosa/Snap4Arduino #306: Distance sensor not working online

draboso, 2021-02-24:
Hi, I'm working on a Chromebook which doesn't work properly on the last app (being unable to disable web security)

so I try it online, using sa5Firmata to work with the hc-sr04, and I get "Inside: Error. This block needs a device running SA5Firmata_tone or SA5Firmata_ir firmware."

Is there anything I could do to avoid the problem?

Thanks a lot,
P.D: I'd swear last year we (some students) could use it on the app

jguille2, 2021-02-24:
Hi @draboso,

Two issues here:

- **About _firmware_**
_Standard firmata_ is not a solution for the ultrasonic sensor features. You need an advanced _firmata_. There are different solutions (look at [our web](http://snap4arduino.rocks/#devices)). And yes, SA5Firmata is a fine solution here. If you don't need IR features, the best solution is **SA5Firmata_tone**.
More info [here](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md)
I guess you know their blocks are into the SA5Firmata library (just preloaded in any current Snap4Arduino version).

- **About chromebooks**
The best option to run Snap4Arduino in Chromebooks i to run it online at [http://snap4arduino.rocks/run/# ](http://snap4arduino.rocks/run/#) using the **Snap4Arduino connector**.
But, be careful!
Maybe you are using the old "Snap4Arduino app". It is not updated and, its biggest problem, is that it can not run JS code and so, more libraries are KO.
Be sure you use the "Snap4Arduino connector". You can find it on the ChromeWebStore [https://chrome.google.com/webstore/detail/snap4arduino-connector/fhmkennllhhclkcjcdaegceilcefedle](https://chrome.google.com/webstore/detail/snap4arduino-connector/fhmkennllhhclkcjcdaegceilcefedle) or you can install it from [our web](http://snap4arduino.rocks/#download).

That's all,

Joan

draboso, 2021-02-25:
Hi, thanks for the answer, but maybe I did not explain myself properly, or I am not understanding something.

I used Snap4Arduino online and the SA5Firmata, but I got this message:
"Inside: Error. This block needs a device running SA5Firmata_tone or SA5Firmata_ir firmware."

thanks

jguille2, 2021-02-27:
Hi @draboso,

Maybe my answer was not very clear... but it's pretty  trying to answer your really concrete question.

- Your error is clear. Is about firmware. And my first part of the answer is the key. [Install SA5Firmata_tone](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md) if you want to use their blocks!
- My second part is about your comments related to the "last app" (on the last app (being unable to disable web security)). Your issue is about firmware, not about the "last app" but it is interesting to understand the situation: "old app" (Snap4Arduino extension) is not updated and needs to disable web security for some libraries) but current app (Snap4Arduino connector) has not these problems.
Joan

closed 2022-12-25 by jguille2