bromagosa/Snap4Arduino #307: Snap4Arduino projects stopped saving

David-Thornburg, 2021-04-21:
As of yesterday, I can no longer save my projects either on the cloud or on my computer. I'm finishing a book on Snap! and Snap4Arduino and am stuck until this gets fixed. I get the following message: 
[image]

Problem shows up on both my Mac and Chromebook.

jguille2, 2021-04-22:
Hi David,

Could you share more details?

I'm testing Desktop and Online versions, and it is running ok (exportint, saving to the cloud...)

Maybe it was a problem with your project? Some Javascript action, breaking the project structure?

Joan

David-Thornburg, 2021-04-22:
I found the problem. I started the project in Snap! then switched to Snap4Arduino. S4A reads Snap! files fine, but doesn't let you save them. I redid the whole project in Snap4Arduino and all is well! This would be fixed if the Arduino blocks library was able to be sent to Snap!

jguille2, 2021-04-22:
Hi David,

But it is not a general issue.
You can open Snap! projects with Snap4Arduino and then, you can export or save them.

I think it was a problem with the Snap! "saving" process, or maybe some "manipulation" of the file. Have you this file? Maybe is corrupted. You can test opening it again with Snap! and see if you have the same issue.

I close this issue because I think it is an one-time problem with a file. 90 / 5000
Translation results
But if you see that it is not so (if you can reproduce it) we continue with the subject to see what happens.

Joan

closed 2021-04-22 by jguille2