bromagosa/Snap4Arduino #308: Logo usage

thaynaNinja, 2021-09-22:
Hello good afternoon.

I would like to know if it is allowed to use the Snap for Arduino logo on a declaration or certificate of course completion for children and teenagers with Snap for Arduino.

jguille2, 2021-09-23:
Hi @thaynaNinja ,

You know all Snap4Arduino is free, licenced under GNU Affero General Public License v3.0. Then, you are able to use everything... and here, the logo itself explains its authorship.

And I like people using Snap4Arduino and Snap!

Just out of curiosity... are you making Moodle badgets for this certification? Or is it a paper certificate or other particular thing?
My question is because there are OpenBadgets... and maybe you can create general Snap4Arduino badgets that can be used anywhere...

Ok,
Joan

thaynaNinja, 2021-09-23:
Hey Joan!!

Thanks for answer me!
No, isn't Moodle badgers. It's a digital certificate.
But thanks for talking about this.

closed 2021-09-23 by jguille2