bromagosa/Snap4Arduino #309: Snap4Arduino, cannot connect to arduino

SRayGT, 2021-09-28:
I am teaching coding using Creator MicroBots and MiniBots 2.0.  I am attempting to block code using Snap4Arduino.  My bot is plugged into my computer USB.  LED lights are on and everything is wired in correctly.  However, when I try to execute code, it says "Error arduino not connected."

SRayGT, 2021-09-28:
I've gone through the steps under the Arduino tab to connect my bot.  It was unsuccessful.  I tried to run the standard firmata and it showed an error message

jguille2, 2021-09-29:
Hi @SRayGT ,

We need more details to help you.

About your first post... it is the normal behavior. If you don't connect any device and play any "Arduinos" block, the message will be just: "Error arduino not connected."

But I guess you have tried to "Connect Arduino". What is the result message after this action?

And some extra comments...
Are you using those devices with other programming languages or utilities?
This question is because some OS (like Windows) need special drivers installed... and others (like Linux) need to set USB (and serial) permissions correctly. For example, if you are using Arduinos, installing Arduino IDE on your computer could fix those problems (because it installs drivers and/or set permissions).

If this is not the problem... Tell us more details (OS, device, error messages...)

Joan

SRayGT, 2021-09-29:
Hi!  Yes I tried to connect on the Arduino tab after attempting to upload
the Standard Firmata on Arduino IDE.  Neither were successful.  I updated
my computer overnight and it finally worked today.  Thanks for your help.

closed 2021-09-29 by SRayGT