bromagosa/Snap4Arduino #311: NWJS Quit unexpectedly

rkoonz2, 2022-06-14:
Hi, I'm working on learning how to program Arduino's. I was told snap4arduino is the new S4A that is no longer supported. (its not 64 bit.) 
Im trying to install Snap4Arduino on my new Mac with M1 chip OS V12.4. (64 bit architecture only) I got past Mac not liking that its not coming from the App Store and now Im getting NWJS quit unexpectedly error when I try to instal it. After downloading Snap4Arduino I open the folder labeled 64 and right click the .app file and then it gives me this crash error.
Am I doing something wrong?

[image]

rkoonz2, 2022-06-20:
I’m guessing from the date of the last posts this is no longer support…

jguille2, 2022-07-09:
Sorry!
Yes, Snap4Arduino is maintained... but I have time this days... and also I have no MAC devices to test and reproduce it.

But I asked tot test in Mac (every release) and I have no other issues about it. I hope other MAC users can help you here?

Anyway, chrome connector could be a solution for you. What's the problem? "Google play store" don't help live outside chromebooks... but you can download the connector from github and install it (you need to change to developers mode and do it).
https://github.com/bromagosa/Snap4Arduino/releases/download/7.0.4/Snap4Arduino_web_connector_crx_7.0.zip

Joan

closed 2022-12-25 by jguille2