bromagosa/Snap4Arduino #316: iMac Sanp4arduino failed to run

danadak, 2022-12-27:
> -------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               nwjs [3246]
Path:                  /private/var/folders/*/Snap4Arduino.app/Contents/MacOS/nwjs
Identifier:            com.nw-builder.snaparduino
Version:               Version 7.0.4 (7.0.4)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-12-27 10:09:33.9932 -0500
OS Version:            macOS 12.6.2 (21G320)
Report Version:        12
Anonymous UUID:        E340F9B8-87D7-D073-FFC1-12221F22D2A9

Sleep/Wake UUID:       30E4426F-D7D0-43D2-A2B3-1BCB74515469

Time Awake Since Boot: 9500 seconds
Time Since Wake:       4547 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff8059ba00e __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff8059f01ff pthread_kill + 263
2   libsystem_c.dylib             	    0x7ff80593bd24 abort + 123
3   libsystem_c.dylib             	    0x7ff80593be92 abort_report_np + 180
4   nwjs                          	       0x10f84b9c8 main + 536
5   nwjs                          	       0x10f84b89d main + 237
6   dyld                          	       0x11f4fd52e start + 462

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x000000011f578600  rcx: 0x00007ff7b06b3768  rdx: 0x0000000000000000
  rdi: 0x0000000000000103  rsi: 0x0000000000000006  rbp: 0x00007ff7b06b3790  rsp: 0x00007ff7b06b3768
   r8: 0x00007ff847143c98   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000246
  r12: 0x0000000000000103  r13: 0x00007f80fa804120  r14: 0x0000000000000006  r15: 0x0000000000000016
  rip: 0x00007ff8059ba00e  rfl: 0x0000000000000246  cr2: 0x000000010f9a5000
  
Logical CPU:     0
Error Code:      0x02000148 
Trap Number:     133

Binary Images:
    0x7ff8059b2000 -     0x7ff8059e9fff libsystem_kernel.dylib (*) <07ad083c-2835-3d18-bb6d-6f081840c8a4> /usr/lib/system/libsystem_kernel.dylib
    0x7ff8059ea000 -     0x7ff8059f5fff libsystem_pthread.dylib (*) <b5454e27-e8c7-3fdb-b77f-714f1e82e70b> /usr/lib/system/libsystem_pthread.dylib
    0x7ff8058ba000 -     0x7ff805942fff libsystem_c.dylib (*) <e42e9d7a-03b4-340b-b61e-dcd45fd4acc0> /usr/lib/system/libsystem_c.dylib
       0x10f84b000 -        0x10f888fff com.nw-builder.snaparduino (Version 7.0.4) <4c4c44d6-5555-3144-a1bb-b4753f1fb10b> /private/var/folders/*/Snap4Arduino.app/Contents/MacOS/nwjs
       0x11f4f8000 -        0x11f563fff dyld (*) <b53641ea-6cd4-39c9-9b14-59c92cfd9584> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
[... 143 lines cut ...]
    }
  ],
  "experiments" : [

  ]
}
}

Model: iMac16,2, BootROM 476.0.0.0.0, 4 processors, Quad-Core Intel Core i5, 3.1 GHz, 8 GB, SMC 2.32f21
Graphics: Intel Iris Pro Graphics 6200, Intel Iris Pro Graphics 6200, Built-In
Display: iMac, 4096 x 2304 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80CE, 0x4B3445364533303445452D45474346202020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80CE, 0x4B3445364533303445452D45474346202020
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x156), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD HTS541010A9E662, 1 TB
USB Device: USB30Bus
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub in Apple Extended USB Keyboard
USB Device: Apple Extended USB Keyboard
USB Device: Apple Optical USB Mouse
Thunderbolt Bus: iMac, Apple Inc., 27.4

jguille2, 2022-12-27:
Hi @danadak  (also I ping @rkoonz2 who had a related issue).

First, sorry but I can't test anything on Macs... because I haven't any apple device :(
Then I need your feedback. The only thing I can say now is that we have a lot of MacVersion downloads (thousands), and only these two issues... So I guess it is running in a lot of Mac devices right now.
Three comments:

1. I see there is the "Error Code: 0x02000148", and it seems that it is related about launching  "unauthorized" or unsigned applications. Maybe the problem is here? Maybe you have to inform to your system that you want to allow them to be installed.
I see some pages talking about "Allow apps downloaded from 'anywhere'", that it is needed... But as I said, I'm not a Mac user and I'm not sure about this.
3. Please. Test this (trying to allow "anywhere apps") with our latest versions (7.0.4 and also with 6.2). If you can't... tell me, but I'm going to release Snap4Arduino8.1 in the coming days. So, if the problem continue... I will want to test with you with that new version... And, if needed, I will look for more information and test with you to try to fix it.
4. In the meantime... You know you can use the online version, don't you?
Just download our Snap4Arduino connector (https://github.com/bromagosa/Snap4Arduino/releases/download/7.0.4/Snap4Arduino_web_connector_crx_7.0.zip) and install it in your Chromium/Chrome/Edge browser (using chrome://extensions and developer mode)
Then, http://snap4arduino.rocks/run will connect with your Arduino-Firmata devices.

Joan

danadak, 2022-12-27:
I myself a new user to Apple. Now its giving me an opening screen I need a subscription.

Not a clue what that about. Steve Jobs still controlling US from the grave.

Regards, Dana.

edgarallan, 2023-01-25:
Hi, 
I've got the same error with mac mini M1 12.6.2 (monterey) and other M1 macs running monterey
I think it is a problem related to unsigned application.
Please let me know
Thanks

jguille2, 2023-01-25:
Hi!
Sorry... and yes, I think unsigned application is related...
I asked for a mac  for testing... but I still don't have it. I hope next week I'll be able to test (and try to fix something in our building). But we'll still be an unsigned ap...because we do not have the capacity to pay to enter the Apple ecosystem.

In the meantime, I guess you know you can run our Snap4Arduino online, don't you?

Joan

jguille2, 2023-01-28:
Hi @danadak and @edgarallan !

https://github.com/bromagosa/Snap4Arduino/issues/317#issuecomment-1407217193

I think the "mac issue" is gone.

Great if you can test and report feedback. Thanks!

Joan

jguille2, 2023-05-25:
Some other tests confirm Mac issue is gone... also in the newest release.
Closing...

closed 2023-05-25 by jguille2