bromagosa/Snap4Arduino #317: Issue with nwjs

Ducks84, 2023-01-11:
Hi everyone,
I'm using a Macbook Pro M1 with MacOS 13.0.1

I can't install snap4arduino, i have a "nwjs" error.

Here's the report:

> -------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               nwjs [81939]
Path:                  /private/var/folders/*/Snap4Arduino.app/Contents/MacOS/nwjs
Identifier:            com.nw-builder.snaparduino
Version:               Version 8.1.0 (8.1.0)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-01-11 21:21:05.9555 +0100
OS Version:            macOS 13.0.1 (22A400)
Report Version:        12
Anonymous UUID:        91774B21-1F74-1BF6-8FFF-945132DAA203

Sleep/Wake UUID:       83373154-3187-4D18-9948-D2BFA9FE4855

Time Awake Since Boot: 33000 seconds
Time Since Wake:       1585 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   ???                           	    0x7ff8ad21e9a8 ???
1   libsystem_kernel.dylib        	    0x7ff81df5f30e __pthread_kill + 10
2   libsystem_pthread.dylib       	    0x7ff81df96f7b pthread_kill + 263
3   libsystem_c.dylib             	    0x7ff81dee0ca5 abort + 123
4   libsystem_c.dylib             	    0x7ff81dee0e13 abort_report_np + 180
5   nwjs                          	       0x104bb2978 0x104bb2000 + 2424
6   nwjs                          	       0x104bb284c main + 236
7   dyld                          	       0x204d8b310 start + 2432

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7fff05750 0x7ff7fff01000 + 18256
1   runtime                       	    0x7ff7fff1166c 0x7ff7fff01000 + 67180
2   runtime                       	    0x7ff7fff1306c 0x7ff7fff01000 + 73836

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x00007fa0fb008800  rsi: 0x0000000000000003  rbp: 0x0000000000000000  rsp: 0x00007ff8615dc258
   r8: 0x0000000204e292c0   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x00007ff8615dc250
  r12: 0x0000000000000103  r13: 0x00000000000000de  r14: 0x0000000000000006  r15: 0x0000000000000016
  rip: <unavailable>       rfl: 0x0000000000000283
 tmp0: 0xffffffffffffffff tmp1: 0x00007ff8ad21e97c tmp2: 0x00007ff81df5f304

Binary Images:
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???
    0x7ff81df57000 -     0x7ff81df90ff7 libsystem_kernel.dylib (*) <0c2fd2c9-777c-3355-b70f-7b1b6e9d1b0b> /usr/lib/system/libsystem_kernel.dylib
    0x7ff81df91000 -     0x7ff81df9cff7 libsystem_pthread.dylib (*) <13b5e252-77d1-31e1-888d-1c5f4426ea87> /usr/lib/system/libsystem_pthread.dylib
    0x7ff81de5f000 -     0x7ff81dee7fff libsystem_c.dylib (*) <5efaf10b-2ec1-32ed-b077-80125e552c8d> /usr/lib/system/libsystem_c.dylib
       0x104bb2000 -        0x104bdcfff com.nw-builder.snaparduino (Version 8.1.0) <4c4c44cd-5555-3144-a1ca-127cbbfb16b4> /private/var/folders/*/Snap4Arduino.app/Contents/MacOS/nwjs
       0x204d85000 -        0x204e1cfff dyld (*) <28fd2071-57f3-3873-87bf-e4f674a82de6> /usr/lib/dyld
    0x7ff7fff01000 -     0x7ff7fff30fff runtime (*) <6f797d84-b330-3656-9a0d-7b3dbd3f8a07> /usr/libexec/rosetta/runtime

External Modification Summary:
  Calls made by other processes targeting this process:
[... 181 lines cut ...]
  "experiments" : [

  ]
}
}

Model: MacBookPro18,1, BootROM 8419.41.10, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  3 2022 02:35:52 version 20.10.965.9.8.7.129 FWID 01-b0e84a9b
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: Laptop, Apple Inc.
Thunderbolt Bus: Laptop, Apple Inc.
Thunderbolt Bus: Laptop, Apple Inc.

I saw someone had this kind of error, but if someone has a solution I will be happy. 
Otherwise I'll use the online version.

Thanks
Ducks84

jguille2, 2023-01-11:
I'm sorry about your issue.

I haven't any Mac device and I can not test it.
I see there are a few more downloads of the osx version and it is the first report.

I ask some friends to test it (to reproduce your issue) and I will fix it as soon as possible.

I believe you are having problems... but I also know that there was the same "error" report by some users on the past version (v7) and there was thousands of downloads and people runing on Macs that version. Then I guess is a configuration issue of the Mac?

Continue.... and we'll find a solution for this.

Joan

jguille2, 2023-01-28:
Hi @Ducks84

I've finally managed to borrow a mac device today... and I've been able to find the problem, fix it and test it :)
https://twitter.com/jguille2/status/1619126474459734017

It is still an unsigned app (you know you have to open Finder and explicitly allow execution... but I think it's fine.
I've also noticed that the first few runs take "too long" to boot...but I'd say that's not a problem.
https://github.com/bromagosa/Snap4Arduino/releases/download/8.1.0/Snap4Arduino_desktop-osx-64_8.1.0.zip

If you can try it, it will be great to have your feedback.

Thanks!
Joan

jguille2, 2023-05-25:
Closing...
Other tests confirm MAC installation is ok (also with the newest version)

closed 2023-05-25 by jguille2