bromagosa/Snap4Arduino #320: The digital reading bricks do not work anymore.

raffoux, 2023-05-12:
Hello,
I had abandoned version 7 because of a malfunction of the arduino bricks (instability). The version 6 works very well, I use it with students from 12 to 16 years old with success. I just tried version 8 and it's worse than version 7, the digital reading bricks don't work at all. I am under ubuntu 22.04 and 20.04 with uno cards directly connected in U.S.B. The analog reading and set digital pin bricks work correctly.
Jean-Christophe

jguille2, 2023-05-13:
Hi Jean-Christophe

I'm sorry for your problems... but we need more details about your issue.

Version 7 is from January 2022 and version 8 is from January 2023. We have had tens of thousands of downloads, and we are also in some educational distributions, in Moodle plugins and directly online. And yours is the first incidence in this sense!

Also to tell you that we have used the application extensively, in schools and workshops with hundreds of students... and we are not aware of this incidence either.
And I'm a Ubuntu user... so it's the most used platform in development testing.

If you want, tell me more details and I'll try to test and reproduce your issues...
- I guess you are using desktop versions (gnu, 64 bits), aren't you?
- You are using UNO boards. Do you use FirmataStandard or are you using other custom firmware?
And in the meantime, I suggest you to try current online version... and also to update your firmware (version 8 has a new feature where you can upload directly you board firmware).

Continue...

Joan

raffoux, 2023-05-14:
Hi Joan,
Indeed, I use a 64 bits version of mate ubuntu. The boards are arduino uno rev 3. Whatever the firmata version used (I tried those provided with version 8), the result is the same. It works perfectly with version 6. I will make other tests.
To be continued ...

jguille2, 2023-05-14:
Hi.

Yes, UbuntuMate64 is also my choice :)

jguille2, 2023-05-14:
I will try tomorrow with a fresh install and we can compare...
- Using desktop version (v8) your user has to be a "dialout" member
- Then, uploading a fresh StandardFirmata to the UNO, just connect and test some electronics/modules

In parallel, you can test the online version and see...

Continue...

raffoux, 2023-05-15:
Hello, 
the results of my few tests are a bit surprising: (whatever the firmata version)

Uno without shield or with a dupont shield, the 6 desktop and 8 online versions work. The 8 desktop version does not work correctly (digital input) with linux or windows.

Uno + grove shield, everything works whatever the version.

The user is a member of the dialout group.
I have no explanation for this.

jguille2, 2023-05-15:
Oh!
I'm not sure (maybe I'm doing something wrong) but in a quick test, I see problems with some pins!!

I have to test this better.... but if you are doing the same... Pins until 7 are working fine and I have problems with digital readings on pins upper 7 (8,9...)

Continue... But maybe there is a bug that no one has reported yet?

raffoux, 2023-05-15:
Indeed, the inputs from 8 onwards work with the desktop version and on the online version it is the opposite, the inputs work up to 7.
I probably used different inputs on each shield.

jguille2, 2023-05-17:
Hi @raffoux 

I've done some tests. It's pretty confusing because there is some problems with an asynchronous process (something changed in the response of  browsers to the Firmata library) and the Snap4Arduino response to this issue is not clear (it depends on the execution).

I have to test more to see what is the best fix (if I change Snap4Arduino code or I fix that Firmata behavior).

In the meantime:

- Online Snap4Arduino doesn't have this problem and all digital readings are running ok.
- And if you want to continue with the desktop version, I add a workaround in this post. Import (unzip the file and import the xml file to your projects)  this block ("Fix digital readings") and run it once at the beginning of your project. The annoying part is that you have to allow JS to do it (in the config menu of Snap4Arduino, just check "Javascript extensions").

And I'll fix this for the next minor release (soon). It's strange for me that nobody else had discover this before. I'm  using regularly the online version (for workshops) and I hadn't seen  it either.

[fixdigitalreadings.xml.zip](https://github.com/bromagosa/Snap4Arduino/files/11500918/fixdigitalreadings.xml.zip)

Joan

raffoux, 2023-05-19:
Hello,
Thank you very much for this work, I will be able to use the 8 desktop version with my students. If I encounter another problem, I promise to tell you about it sooner.
And thanks again, your software is excellent.
Jean-Christophe

jguille2, 2023-05-23:
Hi Jean-Christophe @raffoux 

Today I've released v8.2.4.
I guess it fixes this issue (over desktop and also online versions).

https://twitter.com/jguille2/status/1660817952265101313

I close this issue. Any problem, continue... Thanks!

closed 2023-05-23 by jguille2