bromagosa/Snap4Arduino #321: Dependency management with npm

joelsgp, 2023-05-28:
i.e. package.json instead of a shell script that downloads an ancient zip file

would be much easier for distributors to package too (e.g. https://aur.archlinux.org/packages/snap4arduino)

jguille2, 2023-05-29:
Hi @joelsgp 
Thanks for your report. You are right. But the issue here is that Snap4Arduino is planning to skip node.js and offer a single web version, using firmata under the WebSerial API.

I keep your issue opened to track this. It will be closed when dropped from node.js dependencies. And if we take too long, we reconsider this realignment later.

Joan

jguille2, 2023-12-11:
Closing this...
Version 9 (before Christmas) will be (still) a node built. But we will be working to move all the features to a custom Snap! library (without a desktop app).

closed 2023-12-11 by jguille2