bromagosa/Snap4Arduino #322: S4A icon not shown in taskbar and explorer and ...

mskicker, 2023-06-05:
[image]
[image]

jguille2, 2023-06-06:
Thanks @mskicker 

[Just I added](https://github.com/bromagosa/Snap4Arduino/commit/dfb929d6ef61e7d3518e0a122c70468c1125343a) that ' build.win.icon'  (and the *linux* one) element to fix this after some nodebuilder change.

But I see (thanks to you, I'm a Linux user) it's not quite fine now :(
I will test it (I need to test that this does not break the linux and mac builder)... before the next minor release.

But the plan is to leave this node.js system and so, we will leave all these issues about the desktop editions.

Thanks for reporting and test a solution. I keep this opened until next release fix.

Joan

jguille2, 2023-12-11:
Hi @mskicker
I'm closing this because any change (I've tested different options) is a real good solution for nwbuilder.
I have also read different reports about this issue... and the best option is to do it manually. With the current assembly the installable version of Windows seems to pick up the icons in the different elements (installer, application...) well. We will try to fix the portable version as well, albeit manually, for the next version 9 (I hope it will be released before Christmas)

closed 2023-12-11 by jguille2