bromagosa/Snap4Arduino #323: Font size too small & painful for study without option to increase its size

ak117611, 2023-08-07:
Your s4a software is useless & irritating to eyes coz of uncomfortable font size . When u create settings to increase it , let me know , call me on 9971820630 , my kids & me can start studying it after its done . No people with specs likes to use it as of now

Also check https://github.com/arduino/Arduino/issues/10921

bromagosa, 2023-10-02:
No free software maintainer should ever reply to rude comments, but anyway...

Go to the gear menu, select `Zoom blocks...`

[image]

Make your block text as big or small as you want.

[image]

ak117611, 2023-10-02:
Ohh I lost my job due to your font size .... didn't know your superfast
reply was due to the rudeness contained in Facts

jguille2, 2023-10-02:
Oh!
It was August, we were on vacation... and Snap4Arduino is a free and open source project... We are volunteers.

And also, you were talking about S4A... and I don't know if you were really talking about Snap4Arduino or about the old S4A (Scratch for Arduino). I forgot to answer this, sorry :(

And also, this is github. A place to share and contribute... You are free to use my work (or not)... and you can improve it freely. 

Bernat has shown you the "Zoom blocks" feature... But also, this is a web app... You can play it online and your browser configuration and screen resolution will do their work...
But you can change the browser zoom!!! That's all

Joan

closed 2023-10-02 by jguille2