bromagosa/Snap4Arduino #330: Arduino tab partially translated (at least when selecting the Italian language)

Dosclic98, 2024-09-20:
As the title says the "Arduino" tab is only partially translated from english when selecting the "Italian" language, while in previous versions it was fine.

[image]

jguille2, 2024-09-21:
Sorry!
Snap! changed its translation protocol... and I tried to adapt all Snap4Arduino translations... But I see there is some problem with the Italian one.

I note this problem and I will try to fix it soon.

Joan

jguille2, 2025-01-07:
Hi @Dosclic98 
Now it is fixed
https://github.com/bromagosa/Snap4Arduino/commit/c1bbc0c63f363342cf64496abf7f11d07c0db60f

I will release the new 10 version soon (a few days).
Joan

closed 2025-01-07 by jguille2