bromagosa/Snap4Arduino #333: Arduino Not working in some circumvent like this

mskicker, 2024-10-26:
maybe we need Queue
commands not work when fired together
[image]

jguille2, 2024-10-26:
Yes @mskicker, you are right.
We saw this "firmata issue" before (some messages were lost if we were sending these messages consecutively). Then we added a small delay (really is a "yield" in the schedule) to fix it. So now we can send consecutively Arduino command blocks and no blocks are lost.

But, yes. I can reproduce your issue. If we try to "force" a simultaneous sending, we see the problem again: some commands are lost

We can think other ways to "yield"... but we don't want to slow down our general schedule. Snap4Arduino try to add to Snap! Arduino features... but it wants to be a full Snap! with the same capacity and rendering...

Note that you can workaround easily this situation. You can "yield" manually, or avoid simultaneous messages. These two workarounds are running ok:

A silly manual yield to avoid simultaneity:
[image]
or better (more clear for users)
[image]

And yes, your current code must to run ok...but as I said, I am not clear if we have to force  the schedule further to correct this Firmata issue.

Thanks,
Joan

jguille2, 2025-01-07:
Hi.
Closing this.
Yes I'm sorry... you see Firmata.js project is not well maintained
https://github.com/firmata/firmata.js

But anyway... the future of Snap4Arduino is changing to WebSerial protocol... and also killing Snap4Arduino itself... because with WebSerial it can be a simple web extension for Snap!

closed 2025-01-07 by jguille2