bromagosa/Snap4Arduino #335: BAD BUG :  change language from EN to AR and reverse it again make Snap4Arduino crazy  (BAD BUG ADD to Snap)

mskicker, 2024-11-11:
[image]
--------------
[image]
------------
[image]
------------
and make S4A crazy after some action
[image]

jguille2, 2024-11-11:
Thanks @mskicker 
Some comments here:
- Current Snap4Arduino is Snap 9.2.1... and current Snap is 10.2.2
Some lang issues was fixed and then, "the pending" thing here is to update Snap4Arduino.
I will try to do it soon (sure before Christmas)

- Anyway, testing your issue, I see some issues even pending in our current Snap. I'ts not erasing stage paintings... but it's loosing some information (like pen state (up/down), size...)

I don't close this issue to check this on new release... and I will try to test further and fix that Snap issue.

Joan

mskicker, 2024-11-13:
I check 9.2.1
this snap version work normally in change language about costumes
---------------------------
any language change to ENG make this BUG (I think default value make problem)
[image]

jguille2, 2024-11-13:
Thanks @mskicker ,

Yes, but I saw that other issues ... and then, "the right thing" now is to fix them in current Snap and later, I will synch Snap4Arduino with Snap.

Continue...

Joan

mskicker, 2024-11-16:
bug here created 
[image]

from s4a 7 to 8

mskicker, 2024-11-16:
RANDOMLY FIXED ERROR
I add this line without think about what's happened exactly (like monkey programmers) and this bug fixed.
Please check that is safe solution AND if we have not any side effect. kindly commit that 
[image]

jguille2, 2025-01-07:
Thanks @mskicker.

I think I will change s4a lang system, building together snap and s4a locales... but in the meantime...
- I don't want to change Snap! sources and then I can't apply your workaround.
- But I change our "decorator" to implement English option in the same way to others (this causes is a background error message because that s4a English locale does not exist... but it is not a problem).

https://github.com/bromagosa/Snap4Arduino/commit/53a36d654d5c6329b395bf9b7987c7563f1ea7cc

Thanks