bromagosa/Snap4Arduino #339: Crash on Mac when trying to connect Arduino

thepaqui, 2025-03-16:
I'm having an issue on MacBook Air M1 2020 (although it has also happened on MacBook Pro machines), macOS Sonoma version 14.7.

Snap4Arduino was recently freshly installed on this laptop and some others (version 10.3.6).

The problem is that Snap4Arduino crashes when clicking on the `Connect Arduino` button, whether or not an Arduino is actually hooked up to the laptop. Note that it sometimes immediately crashes, and sometimes the port selection window pops up for a split second before crashing.
Uploading the Firmata SA5 tone through Settings -> More Supported Devices -> Upload crashes immediately upon clicking the Upload button (crashes both the webpage and Snap4Arduino).
Uploading the firmware through regular Google Chrome actually works fine though!

I copy pasted the crash report below (the full one is in the attached text file).

I would greatly appreciate some help, as I am not at all familiar with Apple machines. Thank you!

[snap4arduino_crash-report.txt](https://github.com/user-attachments/files/19271696/snap4arduino_crash-report.txt)

> -------------------------------------
> Translated Report (Full Report Below)
> -------------------------------------
> 
> Process:               nwjs [44048]
> Path:                  /Volumes/VOLUME/*/Snap4Arduino.app/Contents/MacOS/nwjs
> Identifier:            com.nw-builder.snaparduino
> Version:               Version 10.3.6 (10.3.6)
> Code Type:             X86-64 (Translated)
> Parent Process:        launchd [1]
> User ID:               502
> 
> Date/Time:             2025-03-16 12:48:07.3435 +0100
> OS Version:            macOS 14.7 (23H124)
> Report Version:        12
> Anonymous UUID:        B20D7C46-09FA-26D8-83C1-3BEAACFF7D19
> 
> Sleep/Wake UUID:       9FD24339-A988-4F28-8F07-C944F390D22B
> 
> Time Awake Since Boot: 1200000 seconds
> Time Since Wake:       2385 seconds
> 
> System Integrity Protection: enabled
> 
> Notes:
> PC register does not match crashing frame (0x0 vs 0x7FF89A06EA78)
> 
> Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread
> 
> Exception Type:        EXC_CRASH (SIGABRT)
> Exception Codes:       0x0000000000000000, 0x0000000000000000
> 
> Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
> Terminating Process:   nwjs [44048]
> 
> Error Formulating Crash Report:
> PC register does not match crashing frame (0x0 vs 0x7FF89A06EA78)
> 
> Kernel Triage:
> VM - (arg = 0x0) Fault was interrupted
> VM - (arg = 0x0) Fault was interrupted
> 
> 
> Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
> 0   ???                           	    0x7ff89a06ea78 ???
> 1   libsystem_kernel.dylib        	    0x7ff809d7ffea semaphore_wait_trap + 10
> 2   libdispatch.dylib             	    0x7ff809c1f28c _dispatch_sema4_wait + 16
> 3   libdispatch.dylib             	    0x7ff809c1f781 _dispatch_semaphore_wait_slow + 98
> 4   IOBluetooth                   	    0x7ff81ce81a3e __53+[IOBluetoothCoreBluetoothCoordinator sharedInstance]_block_invoke + 538
> 5   libdispatch.dylib             	    0x7ff809c1edbc _dispatch_client_callout + 8
> 6   libdispatch.dylib             	    0x7ff809c2004e _dispatch_once_callout + 20
> 7   IOBluetooth                   	    0x7ff81ce81822 +[IOBluetoothCoreBluetoothCoordinator sharedInstance] + 46
> 8   IOBluetooth                   	    0x7ff81ce2716e -[IOBluetoothDeviceInquiry init] + 190
> 9   IOBluetooth                   	    0x7ff81ce2706c -[IOBluetoothDeviceInquiry initWithDelegate:] + 38
> 10  nwjs Framework                	       0x120dbce5e 0x1199f4000 + 121409118
> 11  nwjs Framework                	       0x120db9e2b 0x1199f4000 + 121396779
[... 699 lines cut ...]
>     0x7ff81ce20000 -     0x7ff81ced8ffb com.apple.Bluetooth (1.0) <8c7e8e09-7255-37d6-9e56-988df689fb30> /System/Library/Frameworks/IOBluetooth.framework/Versions/A/IOBluetooth
>     0x7ff809e21000 -     0x7ff80a2baffc com.apple.CoreFoundation (6.9) <6288edcb-0724-3dc9-b3f7-d1740b8bda64> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
>     0x7ff814d66000 -     0x7ff815000ff3 com.apple.HIToolbox (2.1.1) <5f06f268-b51f-31fc-9dfc-ab1eced819c9> /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
>     0x7ff80d42e000 -     0x7ff80e85eff7 com.apple.AppKit (6.9) <398f1293-8f1b-3ec3-85cd-3292a470d5d5> /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
>     0x7ff8099ec000 -     0x7ff809a2dfc3 libobjc.A.dylib (*) <0952e3af-b235-3efb-b93c-70903420f73f> /usr/lib/libobjc.A.dylib
>     0x7ff809dba000 -     0x7ff809dc5ff7 libsystem_pthread.dylib (*) <79ecab15-71f1-3d6a-8a96-0623e622205f> /usr/lib/system/libsystem_pthread.dylib
>     0x7ff80f3dc000 -     0x7ff80f3f9ffc com.apple.TCC (1.0) <8246b5d6-5a33-3b6c-a695-ff9749ab79a0> /System/Library/PrivateFrameworks/TCC.framework/Versions/A/TCC
>     0x7ff809ac3000 -     0x7ff809b0affd libxpc.dylib (*) <9c604e74-f5e5-3496-8da4-9c093f6beb44> /usr/lib/system/libxpc.dylib
>     0x7ff809dc6000 -     0x7ff809deafff libdyld.dylib (*) <76c5dd13-0bb5-3997-a269-a98694254189> /usr/lib/system/libdyld.dylib
>     0x7ff809c66000 -     0x7ff809cedff7 libsystem_c.dylib (*) <17b641ba-925c-39a9-aa43-ab3b0bcdfe01> /usr/lib/system/libsystem_c.dylib
>     0x7ff80ad52000 -     0x7ff80b988ffb com.apple.Foundation (6.9) <d0176708-70cb-36b3-99a1-3a63581d9bb6> /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
> 
> External Modification Summary:
>   Calls made by other processes targeting this process:
>     task_for_pid: 0
>     thread_create: 0
>     thread_set_state: 0
>   Calls made by this process:
>     task_for_pid: 0
>     thread_create: 0
>     thread_set_state: 0
>   Calls made by all processes on this machine:
>     task_for_pid: 0
>     thread_create: 0
>     thread_set_state: 0

jguille2, 2025-03-16:
Hi @thepaqui 

I'm sorry about these problems.
I'm not an Apple user and I haven't any mac device to test... but I will try to test with other collaborators and also try to find time to reproduce this and fix it.

One question.
You say that uploading is ok with regular Google Chrome. I guess you are testing this using Snap4Arduino online with the same Apple device. Is this right? And then, what about "Connect Arduino" with the same devices using online version with google chrome?

Continue...
Joan

thepaqui, 2025-03-16:
Hi again!

I currently cannot test anything more, but I tried some things after typing this issue and I noticed that there were quite a few references to Bluetooth in this crash report.
So I checked and found out that Google Chrome was the only app on the Mac to have Bluetooth permissions.
Perhaps the problem is with the code that asks for Bluetooth permissions for Snap4Arduino before showing the list of devices?

I unfortunately don't have admin privileges on those computers so I sent a request to the IT department these laptops belong to.
I hope they'll be able to quickly try the setup with Bluetooth permissions granted, and I'll be sure to post an update :)

jguille2, 2025-03-16:
Thanks.
To test online version with the same computers will be great.
We need to test both operations: updating firmware (Settings->Upload firmware on UNO boards->) and Connecting to Arduino main feature.

Continue...

thepaqui, 2025-03-23:
Hello again, sorry for the late reply.

Here's my update : Allowing Snap4Arduino to use Bluetooth manually made both crashes disappear: uploading firmware and connecting the board (on all machines). The online version (on the same computer) with Bluetooth permissions given to Chrome also works great.

All in all, I'm conviced the problem comes from not having the Bluetooth permissions.

jguille2, 2025-08-30:
Hi @thepaqui  and sorry for the delay :(

I'm closing this issue.
There will not be new Snap4Arduino versions because we are presenting (it's just released, with Snap! v11) the new S4A Connector. With it, we can run directly Snap4Arduino features from Snap! (directly without any connector nor installation).

https://www.snapcon.org/conferences/2025/program/proposals/835

Any question, feel free to open another issues, to report problems or ask for something.

Joan

closed 2025-08-30 by jguille2